nRF24/RF24Audio #6: WM8731 codec ( audio codec shield ? )

benbiles, 2015-03-20:
Just wondered how hard it would be to add a codec support like the WM8731 to RF24Audio?
would be nice to have to have 24bit 48k audio streamed from one arduino Due / RF24Audio to
another arduino / RF24Audio? 

I found this https://github.com/delsauce/ArduinoDueHiFi which I think supports the codecs I2S over I2C

might be a nice addon for people that want more than 8bit audio ?

TMRh20, 2015-03-20:
In its current form, probably not worthwhile, since there is no Due/ARM
support, and I have no immediate plans to develop it. I would also suggest
a completely different process, likely using ack payloads, if looking for
high quality one-to-one audio, since the current library mainly
demonstrates the multicasting capabilities of the radios on AVR devices.
Technically, the radio portion of this would be relatively straight forward
with a one-to-one config, (auto-ack = clean, digital audio, minimal
errors,lower transfer rates,   multicasting = higher error rates, lower
quality audio with no error correction for timing, etc. ) however, audio is
complex, so extra features and higher quality audio requires more attention
to the details, and can add complexity to the code.

benbiles, 2015-03-20:
Hi, thanks for your reply.  Does the RF24 library support auto-ack = clean etc? or would I need to write my own RF24 library to achieve that? also do you think a timestamp of some sort would help keep the packets in order?

I guess RF24 library is for AVR and not ARM too.. maybe I should rethink this ! I'm trying to stream high quality audio form A - B with a codec module at each end and RF modules for link. prooving much more complex than I thought ! I'll take another look at the NRF24L01+ data sheet and see if I can learn more there too.

thanks again, ben

benbiles, 2015-03-20:
I found RF24 has auto-ack on by default but I suppose like you say there is more detail in streaming high quality audio. RTP works in a wi-fi network well but has little or no error correction ( essentially UDP )? so what more attention to detail would add complexiy to the code? If the I2S data from codec could be passed direct to the RF24 library and streamed over the link in byte chunks then it would work? is it the buffer, fifo / memory that would be a problem? or clock / sample rate etc ? 
trying to understand what the challenge would be.. any pointers in the right direction would be awsome !! it seams overkill to run an entire OS ( linux ) just to stream digital audio packets fro A - B. I'm trying to make a low power battery application..

TMRh20, 2015-03-20:
RF24 core lib does support ARM devices. To be perfectly honest, I don't
want to get into all the technical aspects, but the main issue with
streaming audio to multiple devices is the timing. There will always be
slight variations in timing between device A and device B, so any audio
streaming needs to deal iwth this. There is usually more than one way, so
the 'best' method is generally dependant on your application requirements
and overall design.

benbiles, 2015-03-20:
Hi, thanks for replying to me. I'll try and learn more about the timing, clocks, sample rates etc and see where that takes me ! someone wrote a library for the mega and codec shield and adjusted the timing for it, maybe I can learn from that..

TMRh20, 2015-03-21:
Np, one of the reasons I suggested using ack payloads is that it can eliminate problems with timing by having the receiver request the data from the transmitter as needed, but is only good for a one-to-one configuration. This is how my original streaming code worked as per the file here https://github.com/TMRh20/TMRpcm/blob/3703e03b2a14c454378802e676bd53695205502b/TMRpcm/pcmRF.cpp but the data-rate is limited to ~48KB/s @ 1MBPS and ~67KB/s @2MBPS
Of course there are other ways of doing things, this is just one option.

benbiles, 2015-03-22:
cool, i'll try and use the radio.writeAckPayload in RF24. I'm busy trying to work out how to split 12 bit Short data from ADC on arduino into low and high bytes to send for now. Thats before I even try to send I2S data from I2C interfaced codec :)

I'm guessing you can only trnasmit byte data with NF24 lib. If I can work that out I'll try and figure out how you pulled the data from transmiter in your code. Sounds like a great idea for non-multicast.

For now it does'nt compile, not sure this is even possible... anyway will post back if I get anywhere..

// read DUE ADC and TX over NF24 as lower & upper bytes? 

void ADC_Handler (void){
  if (ADC->ADC_ISR & ADC_ISR_EOC7){    // initialize when ADC end-of-conversion is flagged (NOTE: pin A0 is AD7, so the EOC flag is on EOC7)
    SigIn = *(ADC->ADC_CDR+7) ;  // get conversion result from the ADC CRD. (NOTE: pin A0 is AD7, so conversion data is held in CRD7)
    
    if (writeloc >= 199) writeloc = 0;  //reset writeloc location to the start of the buffer if it reaches the end
    buff [writeloc] = SigIn;            // assign SigIn to the write location in the buffer
  
  radio.write(buffbytelow [writeloc],8);  //send LOW byte!
  radio.write(buffbytehigh [writeloc],8); //send HIGH byte!
   
   byte low = (buff [writeloc] & 0xFF);
   byte high = (buff [writeloc] >> 8) & 0xFF;
  
   radio.writeAckPayload(1,low,8);    //send low byte over radio ???
   radio.writeAckPayload(1,high,8);   // send high byte over radio ???
   
writeloc++;                         // move writeloc on 1 place in the buffer

closed 2021-11-06 by TMRh20