nRF24/RF24Audio #11: running @8MHz internal oscillator

imayoda, 2016-02-18:
Hi there and thanks in advance for your work.
Is it possible to run this library on 3,3V avr chips @8Mhz internal oscillator?

I guess there's some counter that is clockwise for 16MHz, disabling operation for lowly clocked chips.

Thanks again

TMRh20, 2016-02-20:
Possibly, but it would require some slight code changes.

The value 1,600,000 could be replaced with 800,000 directly, or an F_CPU variable usually contains the CPU speed, so it could be detected automatically.

ie: Change ```ICR1 = 10 * (1600000/SAMPLE_RATE);``` to ```ICR1 = 10 * (800000/SAMPLE_RATE);``` wherever the value 1600000 appears.

imayoda, 2016-02-21:
I have applied the suggested modifications (it appears 3 times plus the oversample setting that was 800000 now 400000) but something else is not right. 
With tx feedback I can hear the speaker rattling , just like a super distorted square wave with something of my voice inside it... nor the led is lighting up.
Any suggestions? maybe there's something else inside rf24 main library? 
Thanks again

TMRh20, 2016-02-22:
You could try changing [this line](https://github.com/TMRh20/RF24Audio/blob/master/RF24Audio.cpp#L562) to ``` ADMUX = (pin & 0x07) | _BV(REFS0) | _BV(REFS1); ``` which will use internal 1.1V reference instead of VCC 3.3v.

Also in userConfig.h you could uncomment ``` #define speakerTX ``` if you haven't already, to send the output directly to the speaker on the transmitter for testing.

imayoda, 2016-02-22:
I get different results in gain, input is good, large 3V excursion in mic's analog pin so that is not the problem.. sounds are good with speakerTX, can hear 8000 vs 16000 samples voice difference.. but still led is constantly off, verified with oscilloscope, pin is grounded.. seems like adc can't cope with pwm on the led timer side.
Is SPI bus autodetected, can't get transmission to the other gemini board (i'm 328p on a breadboard x2)?

TMRh20, 2016-02-23:
SPI speeds should be auto-adjusted by the [arduino spi library](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/SPI/SPI.h) with the new IDE/SPI Transaction functions.

The LED functionality is dependent on the same timer that handles millis() etc. The pin is set to output in begin() and fed values. If you are using the standard IDE etc, it should work, but the CPU freq may throw it off a bit. I'm not sure, but if the speaker output is working, that is a much better sign the radio should work. You can test the radios using the core RF24 library gettingstarted examples to ensure radio functionality, pins etc.

imayoda, 2016-03-11:
I've been off for a while, now everything works with minor modifications to main cpp as you suggested.
It was a faulty nrf module that was bugging my setup..
Now will try to best out quality on a breadboard, because sound is choppy and weird.
Thanks a lot

utsxumiao, 2017-10-28:
It's been a while but I am still interested in how did this project go? @imayoda Did you finally get it working in an acceptable sound quality? Could you share your changes (oppose to TMRh20 standard 16000 version). I am trying to build a pair of Walkie Talkie out of 3.3v Arduino so it can be powered up with single 18650 cell. Thanks! Btw, I'm a big fan of @TMRh20 great work!

imayoda, 2017-10-28:
hi @utsxumiao  ... project never left the breadboard unfortunately .. sound was kinda acceptable but had to go over with too little time left for it.

be1ieve, 2017-11-29:
By simply modify three 1600000 to 800000, the minimal example works on 8mhz pro mini. Not quite sure about the audio quality, might be distorted a little. But human speech is still recognizable.
By the way, huge thanks for your pcm lib and this one. really fun to build something on top of it.

closed 2020-11-11 by TMRh20