nRF24/RF24Audio #20: multi recever

maramb81, 2018-07-17:
Hi Guys 
could i ask, how many receiver could i use with one tx ? 
i want to create a wireless headphone system with about 80 receiver headphone

is it possible ?

thank you

maramb81, 2018-07-17:
we want to create a silent disco party system with cheap headphone, image a disco party without music but just headphone. Standard system use UHF transmission and has three switch to select a kind of music, we want just one tx so you could listen just one dj.

umerkhanfarooq, 2018-07-29:
i am making a system in which i have two transmitters and two receivers but when i start both transmitter and receiver my receivers leds start blinking.transmitters are disturbing my  receivers.......

umerkhanfarooq, 2018-07-29:
[image]
[image]

umerkhanfarooq, 2018-07-29:
or may be my receivers are listening both transmitter.

closed 2020-11-11 by TMRh20