nRF24/RF24Audio #22: RF24Audio with TRRS Audio Jack for use with Headphones?

CzEMC, 2018-10-09:
Hello, I was wondering how this diagram:
[image]
can be modified to instead have the TRRS 3.5mm Jack be utilized for audio output and input.

The jack has four pins: Sleeve, Ring2, Ring1 and Tip. 
[image]
What would I change in the sample diagram to get it so that Audio is outputted to the connected headphones and the mic from those headphones are used as an input when transmitting?

CzEMC, 2018-10-10:
Mic is the Sleeve.
Ring2 is Common Ground.
Ring1 is Right.
Tip is Left.

I assume that the audio coming out of pin 9/10 is mono, so I would be using Right.

However, why does audio IO at pin 9 and 10, why not have the speaker connect to one pin, like pin 9 and the other pin return to ground? 

(Pin 9 -> Speaker -> Arduino Ground) 

I'm confused, as I know that connecting Ring1 and Tip to pins 9 and 10 will not produce the desired outcome as there's no ground/return.

So do I the connect ground (Ring2) to one of those pins, say pin 10 and connect Ring1 (Right) to pin 9?

And for the microphone, I would connect the sleeve to pin A0, and I'm guessing that it will detect sounds by using the electrical resistance to ground?

Thanks in advance for any help and clarification.

payamsbr, 2019-07-10:
[image]

it is what i did for audio jack

closed 2020-11-11 by TMRh20