nRF24/RF24Audio #23: fix it

payamsbr, 2019-07-10:
**in line 580**, you commented that adc auto trigger is set on **TIMER1 Capture interrupt flag**.
it require ADTS[2:0] to set 1 but in the code you just set ADTS0 twice.
please tell if the code is ok and comment need to be updated. tnx

`ADCSRB |= _BV(ADTS0) | _BV(ADTS0) | _BV(ADTS2);           //Attach ADC start to TIMER1 Capture interrupt flag`

TMRh20, 2019-07-11:
I think it should work like it is, but the code should be corrected.

payamsbr, 2019-07-11:
> I think it should work like it is, but the code should be corrected.
> […](#)

i got it, thank you for the response. its a while . its risky to touch any thing when they'r working fine

closed 2019-07-11 by payamsbr