nRF24/RF24Audio #26: Data hum/buzz

andraugust, 2020-01-22:
Hi, Thanks for the library. There is a buzz in the recieved data.  Any ideas why?

The received data in a quiet environment looks like this:
[image]

If I read the mic directly with analogRead (no streaming) there's no hum.

## SETUP

### Transmitter: Similar to examples/GettingStarted.
```c
/*
Transmitter
*/
#include <SPI.h>
#include <RF24.h>
#include <RF24Audio.h>

RF24 radio(7, 8);    // CE, CSN
RF24Audio rfAudio(radio,0);

void setup(){
    radio.begin();
    rfAudio.begin();
}

void loop(){
    rfAudio.transmit();
    delay(1000);
}
```

### Receiver:  Similar to examples/USB_Audio.
```c
/*
Receiver
*/
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

RF24 radio(7, 8);    // CE, CSN
const int PAYLOAD_SZ = 32;            // payload can be 1-32 bytes
const uint64_t addresses[2] = { 0xABCDABCD71LL, 0x544d52687CLL};

void setup(){
    while (!Serial);
        Serial.begin(115200);              // 9600, 14400, 19200, 28800, 31250, 38400, 57600, 115200
    radio.begin();
    radio.setChannel(1);
    radio.setAutoAck(0);
    radio.setCRCLength(RF24_CRC_8);       // Only use 8bit CRC for audio
    //radio.setDataRate(RF24_1MBPS);      // Library default is RF24_1MBPS for RF24 and RF24Audio
    radio.openWritingPipe(addresses[0]);  // Set up reading and writing pipes. 
    radio.openReadingPipe(1,addresses[1]);// All of the radios listen by default to the same multicast pipe  
    radio.printDetails();
    radio.startListening();               // Need to start the radio listening
}

byte data[PAYLOAD_SZ];
void loop(){
    if (radio.available()){
        radio.read(&data, PAYLOAD_SZ);
        for (int i=0; i<PAYLOAD_SZ; i++){
            Serial.println(data[i]);
        }
        Serial.println("");
    }
}
```

### Devices
- [Transmitter and receiver](https://www.amazon.com/WayinTop-NRF24L01-Transceiver-Regulator-Compatible/dp/B07PBBC4H9/ref=pd_sbs_147_8?_encoding=UTF8&pd_rd_i=B07PBBC4H9&pd_rd_r=bd836b6e-e596-4212-b268-9e1fb23c1be7&pd_rd_w=7LH7X&pd_rd_wg=bQjf9&pf_rd_p=bdd201df-734f-454e-883c-73b0d8ccd4c3&pf_rd_r=MK5MBM2WP1Z4E53B5ZVQ&psc=1&refRID=MK5MBM2WP1Z4E53B5ZVQ)
- Transmitter is a nano and receiver is uno, both powered by usb.
- Mic: [Electret Microphone Amplifier - MAX9814 with Auto Gain Control](https://www.adafruit.com/product/1713).

Thanks

TMRh20, 2020-01-22:
Based on my experience I assume the humming sound comes from electronic interference. 

Whether it has to do with the internal timers, SPI etc, I’m not sure, but it is usually reduced by using batteries instead of USB power.

dennisdebel, 2020-03-20:
I tried everything. And solved it using 470uF cap on power line of the rf module, 1000uF on my arduino (nano) and change the sample rate to 20khz and datarate to 240kbps. Now even usb power via computer is fine. Good luck (had the exact same noise fingerprint the OP shows, also with proper psu)

closed 2021-11-06 by TMRh20