nRF24/RF24Audio #27: [Question] what about streaming I2S data?

2bndy5, 2020-10-14:
I've been looking into porting this library to my circuitpython-nrf24l01 library, and I noticed that [many of the platforms that run on CircuitPython firmware natively support I2S](https://circuitpython.readthedocs.io/en/5.3.x/shared-bindings/support_matrix.html#) protocol for digital audio via the builtin [`audiocore` module](https://circuitpython.readthedocs.io/en/5.3.x/shared-bindings/audiocore/__init__.html#module-audiocore). 

So, is there any usefulness to adding an I2S version of the RF24Audio class? Perceptually, I'm imagining just transmitting the I2S data line and leave the WS (Word Select AKA Channel Select) line implied by requiring that RX & TX nRF24L01 nodes have a matching startup configuration (concerning total number of channels, sample rate, and 8-bit vs 16-bit).

See also the [I2S specs](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiI45uWj7PsAhVOvJ4KHVvaBYsQFjAAegQIAxAC&url=https%3A%2F%2Fwww.sparkfun.com%2Fdatasheets%2FBreakoutBoards%2FI2SBUS.pdf&usg=AOvVaw1gEn76-OcByPw5CoM1x8Je)

The RPi -- version 1B+ and later -- needs I2S activated, but it does have a dedicated I2S bus. However, CircuitPython's `audiocore` is not available on Linux SoCs (rightfully so). But I have a feeling this library always had limited compatibility with the RPi due to lack of ADC builtin (concerning real-time audio input) and only 2 hardware-driven PWM GPIO pins (concerning audio output).

TMRh20, 2020-11-11:
This library is mainly designed for AVR devices, and I would protest against expanding its scope. :p

I've written another library [AutoAnalogAudio](https://github.com/TMRh20/AutoAnalogAudio) which is much more versatile and almost has support for the ESP32 and I2S. I could add support for the Arduino I2S lib there, but don't have any supported boards available.

2bndy5, 2020-11-11:
Sounds good.

closed 2020-11-11 by 2bndy5