nRF24/RF24Audio #28: deploy documentation using Github Pages

2bndy5, 2020-11-28:
This issue is part of an effort to start hosting nRF24 org repos' docs on github per individual repo instead of uploading all RF24* lib's docs to tmrh20.github.io.

I've created a github workflow that builds the docs on every commit. This same workflow will upload the generated HTML files to the "gh-pages" branch upon publishing a new release. Notice I've already taken the liberty to create the "gh-pages" branch for this repo.

1. For this workflow to work properly, this repo's settings needs to have the "gh-pages" branch set to the Github Pages Source option:
[image]
    After enabling this option, I expect the first deployment to Github Pages to render an error message.

2. Additionally, it would less visually noisy if the "packages" & "environments" sections are hidden in the repo's sidebar settings:
[image]

When these stipulations have been met in the repo settings I will submit a PR

2bndy5, 2020-11-28:
This repo seems to need a special circumstance because there are no published releases. I could alter the workflow to deploy the docs on every commit (provided doxygen has no problems). @TMRh20 Please advise.

TMRh20, 2020-11-28:
Yeah, I don't expect a lot of commits, so that shouldn't be a problem.

2bndy5, 2020-11-28:
I'm also going through and eliminating as many doxygen warnings as I can. I'll also add a condition to isolate these new workflows to only the master branch.

2bndy5, 2020-11-28:
The only disadvantage to not using releases will be having to bump the PROJECT_NUMBER (in the doxyfile) for each new version of the docs.

closed 2020-12-03 by TMRh20