nRF24/RF24Audio #33: PrivateChannels Not Working?

aflventures, 2021-03-18:
Hi,

I have uploaded the PrivateChannels sketch to 2 radios, first with radio = 0, and second with radio = 1. In this case radio 0 has the audio source to transmit.

However, it looks like radio 1 is always receiving the audio, no matter how I control radio 0 via serial (i.e. typing '2' for radio 2 private channel). Any configuration and it still receives the audio. 

Any ideas?

usama383, 2021-07-03:
I am facing a similar issue as well, the PrivateChannels sketch is working only on transmit and receive cases selection, not for broadcasting. When I placed radio.broadcast command in void loop directly it worked on A1 pit button press transmitting only to the node of that address. 
What I want to achieve is a private communication and broadcast all switching on button press.
Did you somehow get PrivateChannels to work? Couldn't find any help on the internet as well only this thread which is not answered.

TMRh20, 2021-11-06:
I noticed privateChannels is missing a call to Serial.begin() but that is not the issue. The broadcast() function needs to be called while the device is transmitting, so you need to call `rfAudio.transmit();` then you can call the broadcast function to select the radio or radio group that will receive the audio.

closed 2023-04-27 by TMRh20