nRF24/RF24Audio #34: Audio multi broadcast

rachanon0807, 2021-03-29:
Hi,

What I want to do is I would like to have 2 master nodes and one in each master node. There will be 5 more sub nodes.
Is it possible for each sub nodes to communicate across the master node? 

Now I can only communicate with each other within the same group.

2bndy5, 2021-03-29:
> have 2 master nodes and one in each master node

This is confusing. "one (what) in each master node". one secondary master node to each master node?

>There will be 5 more sub nodes.

in relation/connection with what node(s)? you mean 5 sub-nodes to each master node?

Have you read the "Advanced Usage" section (at the bottom) of the [docs' main page](https://nrf24.github.io/RF24Audio/)? Seem like that info might help, but I'm not exactly sure what you're describing.

rachanon0807, 2021-03-29:
I'm sorry 

I would like to create at least two communication groups. And I want all subnodes to be able to communicate with each other.

2bndy5, 2021-03-29:
I think what you want to do isn't possible depending on the behavior of these 2 master nodes. Not because of a RF24Audio limitation, but because of the nRF24L01 transceiver limitation.
- Only 1 transceiver should be transmitting at a time (within a group using the same address).
- Each transceiver can only listen to up to 6 other nodes at a time, but RF24Audio reserves pipes 0 and 1 for designated purposes because those are the only pipes that can have different 5 byte addresses (pipes 2 - 5 borrow their first 4 most significant bytes from pipe 1).

Trying to apply any kind of networking (other than what I already described above) with RF24Audio will only introduce a significant lag and dropped frames. If a radio received signals from 2 separate transmitters simultaneously, then 1 of those signals will be ignored because the transceiver can only decode 1 OTA (over-the-air) transmission at a time (this is often described as collision). RF24Audio disables the transceivers' auto-ack functionality for speed and multicasting reasons, so there's no way to verify what signals got through or what signals got dropped.

rachanon0807, 2021-03-30:
I want 1 transmitter to be able to send data to the rest of the sub nodes.

It may be transmitted simultaneously from time to time. But not all nodes are sent at the same time.

I thank you very much for the information.

2bndy5, 2021-03-30:
did I satisfy your question? can I close this issue?

rachanon0807, 2021-03-30:
Yes you answered my question
Thanks again

closed 2021-03-30 by 2bndy5