nRF24/RF24Audio #38: CT Sensor to Arduino Uno

calla1, 2021-08-18:
I am currently trying to interface my CT Sensor with my Arduino Uno, and it comes with an Audio Jack connector, so I purchased a little TRRS breakout module with Pins named: Sleeve, Ring2, Ring 1, and Tip.
I am wondering what pins on the Arduino I would connect these to, so I can read current from my CT Sensor.  

[image]
[image]
[image]

TMRh20, 2021-08-18:
I think you connect sleeve to GND and the audio input/output to one or two of the others.

calla1, 2021-08-19:
So, the audio input/output (possibly the TIP and either ring1 or ring 2), would they connect to two different Analog pins?

TMRh20, 2021-08-19:
Yup

closed 2021-11-06 by TMRh20