nRF24/RF24Audio #39: NRF24L01 Analog Output for Speaker

AmeerHafez, 2021-09-01:
Hello there. I tried to read the output of a receiver and the signal looked digitized. Is there anyway to output it in analog form?

And yes I tested it with a normal speaker and it sounds fine but the signal I receive on the oscilloscope looked like a pwm signal

Avamander, 2021-09-01:
What are you trying to do? What do you mean by "tried to read"?

AmeerHafez, 2021-09-01:
read as in viewing it on oscilloscope. I'm trying to output analog signal from the receiver side but it looked digitized..

TMRh20, 2021-09-01:
It is PWM. The smaller Arduino devices don’t have a DAC. The best you can do i think is filter the output to get more of an analog signal.

AmeerHafez, 2021-09-01:
mhmm .. if i lower the sample rate of the transmission is it going to make it more like an analog signal? I'm planning to have a similar input signal by avoiding filters or DAC .. so what i understand is that arduino nano/uno can not output analog?

Avamander, 2021-09-01:
> so what i understand is that arduino nano/uno can not output analog?

No, they can't.

AmeerHafez, 2021-09-01:
> It is PWM. The smaller Arduino devices don’t have a DAC. The best you can do i think is filter the output to get more of an analog signal.
> […](#)

Well what kind of filter? and how do I design the filter? Sorry I'm new to this

Avamander, 2021-09-01:
A DAC would work better, but you can use a low-pass filter that lets your intended frequency range trough, but filters the rest. There are calculators out there for low-pass filters.

AmeerHafez, 2021-09-02:
> A DAC would work better, but you can use a low-pass filter that lets your intended frequency range trough, but filters the rest. There are calculators out there for low-pass filters.

mhmm.. is it possible to output the analog signal using arudino duo? It has embedded DAC inside it...

TMRh20, 2021-09-02:
The auto analogue audio library supports Arduino due but is a bit harder to use.

AmeerHafez, 2021-09-11:
Ok thank you.. That is what i needed to know

closed 2021-09-11 by AmeerHafez