nRF24/RF24Audio #40: Nice code, but one problem I can't handle

hbxiscool, 2021-11-05:
Hi @TMRh20

I have used your sketch (Minimal.ino) and my Arduino Walkie Talkie dream Come TURE! Thank you very much!
My problem is that, I have two Walkie Talkies'  Transmit buttons( Pin A1) pushed, of course they are not working now, then I get there, sometimes that one of my Walkie Talkie come into 'silence mode', I try pushing the button and light the led on Arduino mini but the buzz not changes pitch, the other Walkie Talkie has no sound out. If I RESET the  Arduino, its back to normal.
This question has perplexed me for a long time.
Thanks!

TMRh20, 2021-11-06:
Are you using the latest RF24 driver from library manager? 
How long does it usually go before failing?
Are you using one of the stock examples or have you modified the code? Please post your code if modified.

hbxiscool, 2021-11-09:
> Are you using the latest RF24 driver from library manager?
yes, v1.4.1
> How long does it usually go before failing?
test as (frequently but not always):
              Walkie Talkie1          Walkie Talkie2
step1       PTT pushed             PTT released               working
step2       PTT pushed             PTT pushed                not working but ok
step3       PTT pushed             PTT released               working
step4       PTT released           PTT released
step5       PTT pushed             PTT released               working
step6       PTT released           PTT pushed                 working
step7       PTT pushed             PTT released               fail
step8       PTT released           PTT pushed                 working

> Are you using one of the stock examples or have you modified the code? Please post your code if modified.

here the code:
#include <RF24Audio.h>
//#include <printf.h>
#include <SPI.h>
#include <RF24.h>

RF24 radio(7, 8);
RF24Audio rfAudio(radio,1);
void setup(){
  //Serial.begin(115200);    // Enable Arduino serial library
  //printf_begin();               // Radio library uses printf to output debug info  
  //radio.begin();                //  Must start the radio here, only if we want to print debug info
  //radio.printDetails();         // Print the info
  rfAudio.begin();    // Start up the radio and audio libararies
  radio.setChannel(123);
  rfAudio.setVolume(3);
}
void loop(){}

2bndy5, 2021-11-09:
Based on your "steps" chart of usage, it looks like once you changed the direction of communication. it only worked in that direction.
```
              Walkie Talkie1          Walkie Talkie2
step1       PTT pushed             PTT released               working
step2       PTT pushed             PTT pushed                not working but ok
step3       PTT pushed             PTT released               working
step4       PTT released           PTT released
step5       PTT pushed             PTT released               working
step6       PTT released           PTT pushed                 working (<-- changed direction)
step7       PTT pushed             PTT released               fail
step8       PTT released           PTT pushed                 working
```
I don't have any suggestion, just thought I'd share my perception.

TMRh20, 2021-11-09:
Are you running the latest version of rf24? There is an old issue that affected streaming applications that would be fixed by updating.

hbxiscool, 2021-11-10:
yes, RF24 latest vision 1.4.1

TMRh20, 2021-11-11:
I found a bug in RF24Audio that may be causing your problem:
Could you try editing the RF24Audio.cpp file in your Arduino/Libraries/RF24Audio directory and move the stopListening() call up per this: https://github.com/nRF24/RF24Audio/commit/2008d4db861b5062e4adfd026250630ad9671b46#diff-d96c4f3847b1e50e3208564e1519586e2367ee45d6db9abc5426528390b75dd4R538
or
manually install this library straight from github.

hbxiscool, 2021-11-11:
Dear TMRh20,
I fixed the code you mentioned in RF24Audio.cpp, and something strange happened.

step1: power on 
step2: PTT1-->PTT2 fail
step3: PTT2-->PTT1 work  
step4: PTT1-->PTT2 work

which is a regular pattern.

So I think maybe some interference caused by the code in setup() as this:
 
 radio.setChannel(123);
  rfAudio.setVolume(3);

I comment out first one:

  //radio.setChannel(123);
  rfAudio.setVolume(3);

and with power on, both direction works fine, but my old 'one way fault' problem happens randomly. I think maybe the bug have something to do with the Channel.

Regards

TMRh20, 2021-11-11:
I made another bug fix. I'm 98% sure this time it should fix your problem.
See https://github.com/nRF24/RF24Audio/commit/58349b4397feb5895c4fbf02a18b4b730e5dea47
You would need to edit the file per above or reinstall from github as before to try it out.

hbxiscool, 2021-11-12:
Loving you @TMRh20, you are really something. Now the code is perfect!!! 

Regards

closed 2021-11-19 by TMRh20

TMRh20, 2021-12-03:
FYI I put in a better fix for this that keeps the sound quality high, so you may want to update again.