nRF24/RF24Audio #41: Need I fix the code when using ATmega168P with 8MHz clock speed?

hbxiscool, 2021-11-26:
Dear @TMRh20 
Need I do some change in the code when using ATmega168P with 8MHz clock speed?
Regards

TMRh20, 2021-11-26:
You would need to edit the following lines and change 1600000 to 800000, change 800000 to 400000
https://github.com/nRF24/RF24Audio/blob/master/RF24Audio.cpp#L121
https://github.com/nRF24/RF24Audio/blob/master/RF24Audio.cpp#L347
https://github.com/nRF24/RF24Audio/blob/master/RF24Audio.cpp#L349
https://github.com/nRF24/RF24Audio/blob/master/RF24Audio.cpp#L569

I think that's about it, but I have no way to test this.
I'll see about getting this updated so its automatic based on the defined CPU speed.
Please let me know if this works.

hbxiscool, 2021-11-27:
Thank you TMRh20, I'll test later.

Regards

TMRh20, 2021-11-27:
I just updated the code to auto adjust the timers, so you can just re-install/update the library to test it out instead of modifying the files manually.

closed 2023-04-27 by TMRh20