nRF24/RF24Audio #42: Best way to do TX and RX at the same time

chabad360, 2022-01-26:
I'm trying to do some research into this system for a walkie-talkie style system where people can talk over each other (for good reason).

I was thinking the simplest way to it would be to have a pair of nRFs in each transceiver module (there would be networking involved, but that would happen separately over serial i think). But now I'm wondering how many I can use before I run out of room.

The other method I though of would be to remain by default in a receiving state, and use audio-in (with a minimum threshold) as an interrupt (or just switching on the next packet). However that would mean switching to between rx and tx really quickly. Is that reasonable?

2bndy5, 2022-01-26:
This radio is not designed to do TX and RX at the same time. If you look at the datasheet about using ACK payloads, then you'll see that it has to drop out of TX to RX the ACK (then return to TX). This process takes place without the RF24 library's control (the bandit is hardcore into the radio firmware).

I believe RF24Audio does not make use of ACK packets (which are what contain ACK payloads). So, the only course of action available to you is using multiple transceivers attached to the same MCU. And, like you said, Instantaneous behavior about which of the 2 transceivers to use would have to be managed somehow. My main concern would be the SPI bus(s) and multi-threading. 

Off the top of my head, I'd probably recommend a RPi Pico board (or anything using the RP2040) because it has 2 cores and 2 SPI buses. In that case you'd definitely have to use Earl's arduino-pico core to run separate programs on each core.

chabad360, 2022-01-26:
Good to know.

I don't need to ACKing (I think), I'm fine if it's lossy, I just need to be able to do both.

Do you know if there's a practical limit to how many devices I could have going at once before I start having serious congestion issues?
Also, is it practical to include a Device ID and then have a broadcast channel for RX and one for TX and then use that to encode which devices should play said packets?

2bndy5, 2022-01-26:
These questions are more specific to implementation detail. If you want to avoid packet collision, then you could use separate channels (1 for TX, 1 for RX), but that's the only option you have because RF24Audio doesn't use the auto-retry feature which requires the auto-ack feature.

chabad360, 2022-01-26:
thanks. I guess I'll just have to experiment to figure out the rest.

closed 2022-01-26 by chabad360

2bndy5, 2022-01-26:
happy coding!