nRF24/RF24Audio #44: Issue with Private Channels

AmeerHafez, 2022-02-18:
So i have set up two communicating transceivers using broadcast. but when I set up another pair of transceivers there is so much interference :( !.. how to avoid this?

TMRh20, 2022-02-18:
> transmitter 1 is number 0
> receiver 1 is number 1
>
> transmitter 2 is number 100
> receiver 2 is number 101

You would have to change the radio channel on one set of devices. Call radio.setChannel(value 1-126); after calling rfAudio.begin();

TMRh20, 2022-02-18:
> Oh.. I just did that.. that stopped interference. but why would the example have broadcast then? I thought it also changes the designated channel?

The broadcast feature allows you to transmit to individual radio numbers or groups of radios sharing the same number. It’s all using multicast so switching radio channel is not required unless running multiple sets of devices.

AmeerHafez, 2022-02-18:
ahhh ok.. So if i change channels with radio.setChannel for multiple devices I don't need to use broadcast I guess?

TMRh20, 2022-02-18:
Yeah, broadcast is for when using a single transmitter to multiple devices