nRF24/RF24Audio #47: Problems with NRF24L01 mini module

DanieElectronics, 2022-07-01:
I'm working on a wireless walkie talkie with atmega328p 3.3V 8Mhz and I need the circuit board to be as small as possible so I designed my board to use the [NRF24L01 mini](https://m.media-amazon.com/images/I/41O0-NZ8U-L._AC_.jpg) module which is supposedly the same as the [standard module](https://images-na.ssl-images-amazon.com/images/I/611Jr0xNGtL.__AC_SY300_SX300_QL70_ML2_.jpg) but with smaller footprint size.
With the standard module everything worked fine at 1MBPS speed and 24000 sampling rate, but when I switched to the mini module I had to decrease the speed to 250kbps and sampling rate to make it work, also the range decreased a lot to only a few meters.
Is there any way to increase the transmit power by software ? Any hardware modifications for increasing the range ?
Regards.

2bndy5, 2022-07-01:
TX power is very dependent on a stable power supply. This is why we recommend capacitors in parallel to the VCC and GND of the module. If you're going for extremely small, then this might require SMD capacitors. More info in [the RF24 repo's COMMON_ISSUES.md](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md).

Remember, like with the nature of transistors, no 2 are exactly alike (even if from the same assembly line).

TMRh20, 2022-07-01:
> Any hardware modifications for increasing the range ?

You can scrape down the end bit of antenna lead to bare metal and solder on a piece of wire. I think about 31mm is the correct length. That should increase range a bit. I've done this with many of my standard modules.

DanieElectronics, 2022-07-02:
> TX power is very dependent on a stable power supply. This is why we recommend capacitors in parallel to the VCC and GND of the module. If you're going for extremely small, then this might require SMD capacitors

I already placed a 10uF SMD capacitor as close a as possible to the VCC/GND pads, I'll try with bigger values and different voltage regulators next time.

> You can scrape down the end bit of antenna lead to bare metal and solder on a piece of wire. I think about 31mm is the correct length. That should increase range a bit. I've done this with many of my standard modules.

As soon as I get some more modules I'll try this !